mmakaay/esphome-xiaomi_bslamp2 #137: "Bricked" lamp after flashing wrong FW

ChickenFood, 2025-04-24:
Hi @mmakaay 
Thanks for the amazing work!

I have followed the guide up until flashing the lamp's esp32 but got confused and flashed a template configuration from HomeAssistant instead
Template FW didn't connect to wifi and didn't create a failsafe hotspot either
I tried to restore FW but it couldn't get any serial output from the lamp

Is there any way where I could factory reset the esp32 (via physical connection) so I can re-flash it with the example config you have provided?

i am using one of these to flash the FW (https://hub360.com.ng/product/esp-link-v1-0-esp8266-01-wifi-download-module/)... These are viable right?

Also the following is the template config I have used

```
esphome:
  name: test
  friendly_name: TEST

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ****

ota:
  - platform: esphome
    password: ****

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: ***

captive_portal:
```

randybb, 2025-04-24:
esp32 in this lamp is a singlecore with a mac written in eFuse, so a "special" config is needed. This is the reason why this is not working and just bootlooping. You can fix it by flashing it via UART - the steps are described here in the [flashing guide](https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/dev/doc/flashing.md).
That module is just usb to uart, so yes, it will work https://github.com/ldijkman/WT32-ETH01-LAN-8720-RJ45-/blob/main/USB%20to%20ESP-01%20Adapter%20schematic.pdf

And the special config is not so special, basically just the `esp32:` part needs to be changed:
```yaml
esp32:
  board: esp32doit-devkit-v1 # esp32dev would work too
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_custom_mac: true
      ignore_efuse_mac_crc: true
```

ChickenFood, 2025-05-18:
sorry for the late reply
i will give it a try when i am free

will let you know if i can get it to work!

closed 2025-11-24 by mmakaay