mmakaay/esphome-xiaomi_bslamp2 #141: [BUG] Not compiling with ESPHome 2025.7.4

szafran81, 2025-07-28:
I'm getting this while trying to compile using ESPHome 2025.7.4:
```
INFO ESPHome 2025.7.4
INFO Reading configuration /config/esphome/dp-mi-lamp-st-komp.yaml...
INFO Updating https://github.com/mmakaay/esphome-xiaomi_bslamp2@dev
WARNING Using `binary_sensor.BINARY_SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `binary_sensor.binary_sensor_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: xiaomi
WARNING Using `binary_sensor.BINARY_SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `binary_sensor.binary_sensor_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: xiaomi
WARNING Using `sensor.SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `sensor.sensor_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: xiaomi
WARNING Using `text_sensor.TEXT_SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `text_sensor.text_sensor_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: xiaomi
INFO Generating C++ source...
INFO Compiling app...
Processing dp-mi-lamp-st-komp (board: esp32doit-devkit-v1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
No dependencies
Compiling .pioenvs/dp-mi-lamp-st-komp/src/main.cpp.o
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/driver/libdriver.a
Compiling .pioenvs/dp-mi-lamp-st-komp/efuse/esp32/esp_efuse_table.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/efuse/esp32/esp_efuse_fields.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/efuse/esp32/esp_efuse_utility.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/efuse/src/esp_efuse_api.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/efuse/src/esp_efuse_fields.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/efuse/src/esp_efuse_utility.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/efuse/src/efuse_controller/keys/without_key_purposes/three_key_blocks/esp_efuse_api_key.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/efuse/src/esp_efuse_startup.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp-tls/esp_tls.c.o
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/efuse/libefuse.a
Compiling .pioenvs/dp-mi-lamp-st-komp/esp-tls/esp-tls-crypto/esp_tls_crypto.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp-tls/esp_tls_error_capture.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp-tls/esp_tls_platform_port.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp-tls/esp_tls_mbedtls.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_adc/adc_oneshot.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_adc/adc_common.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_adc/adc_cali.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_adc/adc_cali_curve_fitting.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_adc/deprecated/esp_adc_cal_common_legacy.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_adc/adc_continuous.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_adc/esp32/adc_dma.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_adc/esp32/adc_cali_line_fitting.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_adc/deprecated/esp32/esp_adc_cal_legacy.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_app_format/esp_app_desc.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_bootloader_format/esp_bootloader_desc.c.o
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/esp_bootloader_format/libesp_bootloader_format.a
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_coex/esp32/esp_coex_adapter.c.o
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/esp_app_format/libesp_app_format.a
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/esp-tls/libesp-tls.a
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_coex/src/coexist_debug_diagram.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_coex/src/coexist_debug.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_common/src/esp_err_to_name.c.o
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/esp_coex/libesp_coex.a
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/esp_adc/libesp_adc.a
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_cam/esp_cam_ctlr.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_cam/dvp_share_ctrl.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_dac/dac_oneshot.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_dac/dac_cosine.c.o
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/esp_driver_cam/libesp_driver_cam.a
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_dac/dac_continuous.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_dac/dac_common.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_dac/esp32/dac_dma.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_gpio/src/gpio.c.o
[... 175 lines cut ...]
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_mcpwm/src/mcpwm_fault.c.o
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/esp_driver_i2c/libesp_driver_i2c.a
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_mcpwm/src/mcpwm_gen.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_mcpwm/src/mcpwm_oper.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_mcpwm/src/mcpwm_sync.c.o
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/esp_driver_i2s/libesp_driver_i2s.a
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_mcpwm/src/mcpwm_timer.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_pcnt/src/pulse_cnt.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_rmt/src/rmt_common.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_rmt/src/rmt_encoder.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_rmt/src/rmt_rx.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_rmt/src/rmt_tx.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_sdio/src/sdio_slave.c.o
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/esp_driver_ledc/libesp_driver_ledc.a
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_sdm/src/sdm.c.o
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/esp_driver_mcpwm/libesp_driver_mcpwm.a
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_sdmmc/src/sdmmc_transaction.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_sdmmc/src/sdmmc_host.c.o
Archiving .pioenvs/dp-mi-lamp-st-komp/esp-idf/esp_driver_pcnt/libesp_driver_pcnt.a
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_sdspi/src/sdspi_crc.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_sdspi/src/sdspi_host.c.o
Compiling .pioenvs/dp-mi-lamp-st-komp/esp_driver_sdspi/src/sdspi_transaction.c.o
*** [.pioenvs/dp-mi-lamp-st-komp/src/main.cpp.o] Error 1
========================== [FAILED] Took 7.69 seconds ==========================
```

BlairC1, 2025-07-28:
I'd recommend waiting for mmakaay to update but I have a working build if you want to try. Just update your config to point at https://github.com/BlairC1/esphome-xiaomi_bslamp2

Remember to switch this back later as this is my play environment.

mmakaay, 2025-07-29:
This has been fixed in the release that I did today, with a fix as provided by @BlairC1.
See: https://github.com/mmakaay/esphome-xiaomi_bslamp2/releases/tag/2025.7.0

closed 2025-07-29 by mmakaay

szafran81, 2025-07-30:
I've made a second clean build and now the MAC errors are gone. But there is still something wrong. HA can't interact with the lamp, and I can't see it on the routers DHCP list (it's weird, but it looks like it doesn't get the IP from the router, but I still can upload a new FW to it).